rhodey/sun-stone #5: choice of IMU

rhodey, 2020-12-05:
I ordered the ICM-20948 from Sparkfun for $16.95 and the magnetometer appeared to be broken: https://www.sparkfun.com/products/15335

Next I tried FXOS8700 + FXAS21002 from Adafruit and was able to get the magnetometer working but have not yet tested the Accelerometer or Gyroscope: https://www.adafruit.com/product/3463

I would like to order another ICM-20948 and have it work because it is a one chip solution, on Digikey the chips are $7.31 a piece and $4 in orders of 100:

https://www.digikey.com/en/products/detail/tdk-invensense/ICM-20948/6623535

rhodey, 2020-12-06:
@miloh if you want to prototype some before we have our first batch of boards ordered I would order the ICM-20948 breakout from Sparkfun and the "ChronoDot" DS3231 RTC from Adafruit:

https://www.sparkfun.com/products/15335
https://www.adafruit.com/product/255

The RGB LED I am suggesting is cheaper on Ebay but at $0.95 you might want to throw a couple in your Sparkfun order:

https://www.sparkfun.com/products/11679

Also if you want to evaluate the two photo-transistor options as I plan to they are only $0.95 a piece:

https://www.sparkfun.com/products/246
https://www.adafruit.com/product/2831

If you find an 9dof IMU besides the ICM-20948 which is similar price or cheaper with example code it would be good to have another option as backup because the first ICM-20948 breakout I ordered from Sparkfun was defective but I am ordering a second tonight.

rhodey, 2021-01-01:
I received another ICM-20948 breakout board from Sparkfun and this time was able to get both the I2C and SPI interface working however there is a terrible amount of drift which I have not been able to correct for. This guy [kriswiner](https://github.com/kriswiner) appears to be the IMU expert on GitHub and in [this issue](https://github.com/kriswiner/MPU6050/issues/6#issuecomment-677826565) he states:

> ICM20984 is not a low noise sensor.

Additionally I discovered that the ICM-20948 uses `1.8v` logic and would require the addition of level shifters to the BOM, now I have given up on this sensor. IMO the next candidate IMU is the [LSM9DS1](https://www.digikey.com/en/products/detail/stmicroelectronics/LSM9DS1TR/4988083?s=N4IgTCBcDaIDYGcC2BOAJggjCAugXyA), the sensor has a good reputation with [kriswiner](https://github.com/kriswiner/LSM9DS1/) and Adafruit has example code for calibrating the LSM9DS1 with MotionCal. However, the LSM is only available on DigiKey in "24LGA" package which looks like it could be a pain to solder DIY.

@miloh can you share any advice for working with "24LGA" package? Is it worth a try?

miloh, 2021-01-01:
@rhodey 
Even for prototyping, I think home/diy reflow with griddle or toaster is the only way to go with these packages.
If the noise on the original can't be reduced -- think it's worth a try because there is no alternative.

rhodey, 2021-01-08:
Thank you @miloh, I received a LSM9DS1 breakout board from Adafruit and found that the performance was **MUCH** better than the ICM20948 and comparable to the original FXOS8700 + FXAS21002 breakout I started with. I have updated the BOM and codebase to use LSM9DS1.

resolved by 9ccb17b449d269a898b00abaceba71a4d003e664

closed 2021-01-08 by rhodey